vitoplantamura/MagicTrackpad2ForWindows #1: Haptic feedback click on simple tap

provalinf, 2024-06-11:
Hi vito,  
It's great that you've been able to take up and improve on the work that's already been done to create this version, I installed it yesterday and I still don't have enough hindsight to give relevant feedback, but it's much appreciated and much more usable, I had tried to build the project under VS some time ago but without too much success.

Do you know if it's possible with existing hardware to get haptic feedback with a simple tap, for a possible next version, or if the “physical” click is required?

Thanks !!!

vitoplantamura, 2024-06-11:
Thank you for your interest in this project! Let me know as soon as you
have feedback of any kind. I have been using this driver for more than a
month (exclusively, that is, I only use the Magic Trackpad all day) and I
am very happy with it.

Regarding your question, I believe it can be done by putting the MT2 in
host click mode. Some work on this has been done in the Linux community (as
usual)... I'll take a look at it in the future but I can't tell you when...

Thanks, Vito

canaryGrapher, 2025-01-03:
Are we working on this feature?