vitoplantamura/MagicTrackpad2ForWindows #3: [Feature Request] Disable the two finger back/forward gesture

CV8R, 2024-06-21:
Would it be possible to add controls in the control panel to disable the two finger gesture for swipe back and forward.  I am regularly, accidentally, going back on the browser.  Usually after completing some long winded form...

Generally though this is fantastic. Thank you for the work!

vitoplantamura, 2024-06-22:
hi,

unfortunately this cannot be done by the driver, since that type of gesture
is managed entirely by the browser.

I personally use FireFox, precisely to solve this type of problems,
considering its high level of configurability.

For example, maybe this could solve your problem:

https://old.reddit.com/r/firefox/comments/138hg87/how_can_i_disable_swipe_left_on_laptop_trackpad/

Vito

CV8R, 2024-06-22:
Hi Vito,

I actually use Firefox too so this could be of help.  I raised the FR because I see in the image for Trackpad++ under two finger gestures section a check box for "Swipe for Back / Forward Navigation" and assumed it was a function of the driver.

[http://trackpad.forbootcamp.org/images/image1_big.png](url)

vitoplantamura, 2024-06-24:
Trackpad++, like Magic Utilities, does not use a Precision Driver.

So it's reasonable to assume that they read the trackpad's raw input and
thus simulate the input of a traditional mouse. In this setup, Trackpad++
can detect that 2 fingers gesture and then simulate the pressing of the
"back" or "forward" button of a virtual mouse.

In the case of a Precision Driver, this is done by the OS or directly by
the application.

Vito

CV8R, 2024-06-24:
Makes sense.  I have made the config change in Firefox and that has it sorted in the browser.

closed 2024-06-30 by CV8R