vitoplantamura/MagicTrackpad2ForWindows #4: Clarification on Bluetooth Support and Future Plans for Magic Trackpad 2

Tosko4, 2024-08-03:
Hi,

Thank you for your hard work on this project—it's been incredibly helpful! I noticed that the readme states Bluetooth is not supported and that the project is designed specifically for USB connections. Could you please clarify if this limitation applies to the entire driver or just the control panel functionality?

I'm trying to decide whether to use this project or another one if I want to use the Magic Trackpad 2 over Bluetooth. Are there any plans to support Bluetooth in the future, or should I look into alternatives?

Thanks in advance for your guidance!

vitoplantamura, 2024-08-05:
hi,

if you install my driver following the instructions in the main README of
this project, using the MT2 via bluetooth will work but the feature set
will be limited to that of the original imbushuo driver. In this case you
will lose the ability to set all the options present in the Control Panel,
such as haptic feedback and all the options to improve the precision of the
pointer.

Although also supporting the Bluetooth version of the driver would be
technically simple, the problem is signing the Bluetooth driver (necessary
for installation on end users' PCs): the cost is 500/600 USD per year and
sustaining this expense in case of my driver would only make sense (even
keeping it free) if thousands of users were using it...

Vito

Tosko4, 2024-08-07:
Understood. Too bad that it costs money. Would there be any way to work with some sort of self signed certificate?

hutchx86, 2024-08-16:
Hi, first of all: love the work you put into this! 
Here's my thought: As far as I remember (from installing a bunch of modded drivers in the past) you only have to disable driver signature enforcement once, install the driver, and then you can reboot and the driver should stay installed, no?

In that case, you could give people the option. Either way, I'll gladly test it, should you ever decide to actually make your tool work with the trackpad over bluetooth as well. :D

hutchx86, 2024-08-16:
After reading a bit more, that might only work for UMDF drivers, and might clash with secure boot/core isolation anyway. Bummer. I'd still be willing to try it, though. lol

NewsGuyTor, 2025-01-10:
> hi, if you install my driver following the instructions in the main README of this project, using the MT2 via bluetooth will work but the feature set will be limited to that of the original imbushuo [...]

I think you could get it way cheaper than that. From what I've seen it would be 300-400 USD for forever. The certificate would only be valid for one year,  but if timestamped the driver would still work when the cert expired.

Still though, it's a lot of money.