vitoplantamura/MagicTrackpad2ForWindows #6: Installation on Windows on ARM devices

reagag, 2024-08-22:
Hi!
I've been using your driver on my x86 laptop for around a month and it has been working great.
Recently switched over to one of the newer Snapdragon X ARM laptops and hit a roadblock trying to install the driver.

I'm able to install the imbushuo drivers no problem, but am hitting a roadblock making changes via your .exe .
Getting the error: C:\Windows\System32\DriverStore\FlleRepository\amtptpdevice.inf_**amd64**_5de6239780ba286e\AmtPtpDevicelJsblJm.dll" doesn't exist.

When I checked the files, I only have a folder called: "C:\Windows\System32\DriverStore\FlleRepository\amtptpdevice.inf_**arm64**_5de6239780ba286e\AmtPtpDevicelJsblJm.dll" 
With the key difference being the actual file being labelled with the arm64, not amd64.

Do you know what to do in this case? Thanks!

vitoplantamura, 2024-08-22:
hi,

I know that the signing requirements for UMDF drivers are stricter on
Windows on ARM.

Unfortunately I don't have a system with Windows on ARM available, and
therefore I can't do a test.

If I sent you a build of the driver for ARM64, could you do a test? It
involves replacing that file that you indicated in your message with the
file that I would send you (first modifying the owner and ACL of the
original file; I would send you all the steps, it's not difficult to do).

Thanks, Vito

reagag, 2024-08-22:
Yeah, that would be great! I originally tried just renaming the file,  but
quickly realized that that's not how it works haha.

vitoplantamura, 2024-08-23:
so the goal is to replace this file:

C:\Windows\System32\DriverStore\FileRepository\amtptpdevice.inf_arm64_5de6239780ba286e\AmtPtpDeviceUsbUm.dll

with the one attached to this message.

If you try to overwrite that file in the Windows Driver Store right away, you should get an error. To enable copying, follow these steps:

1) Disconnect the MT2 from the USB port.

2) set the file owner to your user (i.e. the user with whom you are logged in to Windows).

3) give your user FullControl permission to that file.

4) try to overwrite the file again.

Let me know,

Thank you, Vito

[AmtPtpDeviceUsbUm_ARM64_20240823.zip](https://github.com/user-attachments/files/16724498/AmtPtpDeviceUsbUm_ARM64_20240823.zip)

reagag, 2024-08-24:
Is the arm driver the one made by imbushuo? I currently have that one installed. Even after replacing it with the build that you provided, the same error still pops up.

C:\Windows\System32\DriverStore\FlleRepository\amtptpdevice.inf_**amd64**_5de6239780ba286e\AmtPtpDevicelJsblJm.dll" doesn't exist.

I think the issue is that the .exe file is trying to target the **amd64** version of the folder, which doesn't exist on my computer since it is installing the driver under the folder titled with **arm64** in it.

Maybe need to rewrite the .exe to target the arm64 version instead?

vitoplantamura, 2024-08-24:
after following my instructions in the previous message, to check if the
procedure worked, you should start the Control Panel and then try for
example to change the haptic feedback settings (and at the end click the
"Apply" button!) .

If you can change the MT2's haptic feedback settings in the Control Panel
(which is not possible with the imbushuo driver), the installation is
successful. Do not click the "Install Driver" button in the Control Panel,
which is the only part of the software that distinguishes between AMD64 and
ARM64.

Vito

vivlim, 2024-08-29:
That worked for me on my Surface Pro 11. Everything seems to be working and the trackpad feels fantastic to use. Nice work!

I'll document the steps I took to get it set up since the permissions ui is a bit fiddly.

1. Download [imbusho's driver](https://github.com/imbushuo/mac-precision-touchpad), specifically [the arm64 version on this release](https://github.com/imbushuo/mac-precision-touchpad/releases/tag/2105-3979)
2. Unzip the contents.
3. Right click `drivers\arm64\AmtPtpDevice.inf` and click "Install" (and click through the dialog that pops up)
4. Go to `C:\Windows\System32\DriverStore\FileRepository\amtptpdevice.inf_arm64_fd8449c22cc620c8` in explorer
5. Right click `AmtPtpDeviceUsbUm.dll` and click Properties.
6. Switch to the Security tab. It should look like this:
[image]
7. Click the Advanced button at the bottom
8. In this dialog, click the 'Change' link near the top
[image]
9. Type in your username and press enter. The 'Advanced Security Settings for AmtPtpDeviceUsbUm.dll' dialog should now list you as the owner, instead of SYSTEM.
10. Press OK.
11. Now the 'Edit...' button in the Security tab should look like this. It doesn't have the shield icon, indicating that your user owns the file and can directly change its permissions.
[image]
12. Click "Add", type your username again, and press enter.
13. Select your user and tick all the boxes in the allow column. It should look like this when you're done:
[image]
14. Click OK twice to close the properties.
15. Now download and extract the zip vitoplantamura shared earlier in the thread: [AmtPtpDeviceUsbUm_ARM64_20240823.zip](https://github.com/user-attachments/files/16724498/AmtPtpDeviceUsbUm_ARM64_20240823.zip)
16. Overwrite `C:\Windows\System32\DriverStore\FileRepository\amtptpdevice.inf_arm64_fd8449c22cc620c8\AmtPtpDeviceUsbUm.dll` with the file from the zip.
17. Plug in the MT2.
18. If you don't already have it, download `AmtPtpControlPanel.exe` and open it.
19. The options in the control panel should affect the trackpad after clicking apply. I turned on "maximum haptic feedback" which made it very obvious that the control panel is working. (it *is* very loud!)

vitoplantamura, 2024-08-30:
hey @vivlim,

Thank you for taking the time to do this test and for writing this guide.

Now that I know that the installation procedure also works on ARM64, I can
modify the Control Panel accordingly.

In the meantime people can use your guide!

Thank you, Vito