vitoplantamura/MagicTrackpad2ForWindows #7: vulnerability fixed?

DEPazifist, 2024-08-23:
Hello, I have found out that you have resumed the project from  [imbushuo](https://github.com/imbushuo) and have also adapted the code somewhat. Unfortunately, I don't understand much about driver programming but I have compare the code and I´m unsure if you alredy addressed the issue described in https://github.com/imbushuo/mac-precision-touchpad/issues/544.
The advise from SAST Scan was to replace  `sizeof(DeviceContext->DeviceInfo->um_size)` to the actual data size expected in the buffer.
Is that right that you do it with `CONST BYTE	mt2Click[]` and `sizeof(mt2Click)`?
Best regards

vitoplantamura, 2024-08-30:
hi,

I already noticed that bug some time ago. imbushuo, during his tests, also
noticed the problem and left this comment just below that code:

// Behavior mismatch: Actual device does not transfer bytes as expected (in
length)
// So we do not check um_size as a temporary workaround.

The bug is still there. I didn't fix it because I thought it was a harmless
bug.

Vito

DEPazifist, 2024-09-09:
Thank you very much for the feedback.

closed 2024-09-09 by DEPazifist