vitoplantamura/MagicTrackpad2ForWindows #8: Sensitivity adjustment and better description of finger lifting

Ilosariph, 2024-09-23:
Hey,
thanks a lot for this project. I love the firm but silent click.
I have two things I'd love to have added.
The first is sensitivity adjustment. I know that the driver has this on it's roadmap, but the project seems to be pretty dead. Is this something that could be added easily?

The second thing is, that I don't understand what the "When you lift your finger from the trackpad" setting does. I tried all three settings but couldn't notice any difference. Is there any way you could add a better description of what this does?

Thanks a lot

vitoplantamura, 2024-10-02:
sorry for the late reply.

By "sensitivity adjustment", excluding pointer speed and acceleration, I
mean what is described in this issue in the original imbushuo repo:
https://github.com/imbushuo/mac-precision-touchpad/issues/405 , and that's
what the "when you lift your finger from the trackpad" setting would like
to address.

To understand what the "when you lift your finger from the trackpad"
setting does, try specifying a pressure of 5 or 10 units :-)

Vito

MoonlitChris, 2024-10-02:
Awesome work on this project but is there an updated preview or release I can download? The one from 2021 is pretty outdated now.

Ilosariph, 2024-10-02:
@vitoplantamura thanks for your reply.
By sensitivity I meant pointer speed. I thought this was something the trackpad controlled, like the DPI on a mouse. There is a setting in the windows trackpad menu for that.
I'll try the setting with a higher value once I get some time, thanks. Feel free to close the issue.

1Revenger1, 2024-10-06:
The driver only passes finger position to Window's precision touchpad stack. The windows trackpad settings are used to change pointer speed.

Are the "When you lift your finger from the touchpad" settings needed? The pointer already stops when the finger is lifted when the "contact" bit is unset (though that requires the "Ignore finger input when not touching the surface" setting to be enabled here)

vitoplantamura, 2024-10-07:
hi @1Revenger1,

it's nice to see you here!

When I examined your driver's trace logs (those generated by the
AmtPtpServiceTouchInputInterruptType5 function), they showed that when you
lifted your finger from the trackpad surface, the TipSwitch value became 0
just 1, 2, or even 3 calls after the Pressure value became 0. This meant,
in practical terms, that you saw the pointer shake for an instant, reducing
the precision of the next tap (with "tap to click" active). That's why I
added those options :-)

Furthermore, after several tests, I decided that I would NOT use the
Confidence value to stop the pointer, since I wanted the greatest possible
precision when lifting the finger from the trackpad.

Vito