vitoplantamura/MagicTrackpad2ForWindows #9: Problems in figma (UI design software)

1478976213, 2024-10-10:
When I drag the canvas, it is recognized as an offset up and down/left and right, and cannot be pushed freely. When I touch and drag again, it can be dragged freely normally. What is the reason for this problem?

vitoplantamura, 2024-10-11:
hi,

Try disabling the "Ignore input from the finger used to click the force
touch button" option.

Also try setting "When you lift your finger from the trackpad" to "Do
nothing".

Vito

1478976213, 2024-10-11:
https://github.com/user-attachments/assets/1eb8c9f8-d4bb-4851-b04f-4290952ea1a6

[image]
After I tried the method you mentioned, I still had this problem no matter how I checked the app. I zoomed in on the web page and then scrolled and still couldn't drag it freely. I don't think it was a problem with the figma program.

1478976213, 2024-10-11:
I suspect there is a problem with the driver of my laptop, and the touchpad of my laptop also has the same problem.

arvidbjorkstrom, 2024-10-17:
It feels like there are two "modes" (this behaviour is both for my magic trackpad and my built in trackpad). If I start scrolling exactly straigth up/down or right/left I'm locked in that direction. But if I start scrolling diagonally the scroll is free.