vitoplantamura/MagicTrackpad2ForWindows #10: Support for 3 and 4 finger tap

htryban, 2024-11-01:
Support for the 3 and 4 finger tap gesture would be fantastic. None of these similar drivers I have found support this, and it would fix my primary lingering frustration about the Apple touchpad on windows. Allowing the gesture to be recognized and passed through to the Windows trackpad manager to support the gestures set in windows settings would be an incredible addition.

hutchx86, 2024-11-25:
Those gestures work fine for me with this driver. I set them up in the Windows settings app

htryban, 2024-11-26:
Did you have to alter any additional settings? I am able to use 3 and 4 finger swiping gestures just fine, but the tapping does not register at all so far as I can tell. I have both taps configured in the Windows Touchpad Settings.

I've tried both wired and Bluetooth on a Magic Trackpad 2 and 3. I am on Windows 11.

hutchx86, 2024-12-07:
Hi! Sorry just saw this now. No, didn't have to change any other settings. Just selected the gestures I want in the settings app and it just works

htryban, 2024-12-07:
Just tried to install on a newly formatted PC and can confirm it does work. So surely on my other computer where it has issues I am running into some conflict with presumably other software I have tried in the past to achieve the same behavior. Will look into fully clearing associate drivers and hopefully come back with a solution that isn't just wiping the PC.

closed 2024-12-07 by htryban

htryban, 2024-12-08:
I figured out my issue. In short, my PowerToys Keyboard Manager settings were overriding and invalidating the shortcut being sent from my tap gestures. 

-------------

I started by removing all touchpad drivers and attempting to reinstall from scratch. Probably unnecessary for my issue, and I probably was a bit overzealous in my removal, but a good thing to start with nonetheless. 

In the control panel for this app, I realized the tap gestures were working, and discovered that when the focused program was running as admin the tap gestures worked properly. 

So I looked into my PowerToys Keyboard Manager settings, where I had remapped my left alt and control keys to one another to mimic MacOS, and made the connection.

I originally assumed that some of the preexisting gestures in the Windows setting touchpad were standalone commands, but as it turns out, they are seemingly all just obfuscated key combination shortcuts. For instance, the gesture I most need is three finger tapping for middle mouse button, which is performed by executing `Win (Left) + Ctrl (Left) + Shift (Left) + F22`, which was in turn being relayed as `Win (Left) + Alt (Left) + Shift (Left) + F22`, which doesn't do anything so far as I can tell. 

I was able to get around this and keep my Keyboard Manager remappings by adding to the 'shortcuts' section of Keyboard Manager and telling it to intercept the `Win (Left) + Alt (Left) + Shift (Left) + F22` and turn it back into  `Win (Left) + Ctrl (Left) + Shift (Left) + F22`.

My Keyboard Manager setup for this now looks as shown below, with the first shortcut being the middle mouse button click and the second being notification center. The shortcut being executed can be discovered by setting it in the Windows Touchpad gesture settings, then while creating a new shortcut click the pencil to select the shortcut keys, and execute the gesture. It looks like this can be done for pretty much any windows precision touchpad gesture, and set on a per app basis via the shortcuts in Keyboard Manager, opening up some interesting possibilities for a ton of gesture customizability similar to a Better Touch Tool on MacOS, which is what I've been trying to replace in windows. 

[image]

hutchx86, 2024-12-10:
Glad you managed to figure this out! And this is good info for other people to have. :)