vitoplantamura/MagicTrackpad2ForWindows #17: Bluetooth Support for Type C?

PythonOrC, 2025-02-10:
Hello,

First of all, thank you so much for your work. It is truly amazing. I was able to get the USB connection working perfectly, but I was just wondering if there is any way to get the Bluetooth functionality. I understand in #4 you mentioned that it is possible to get the imbushuo driver working with Bluetooth. Does this apply to the Type-C version as well? I tried the following steps:

1. Disconnect USB connection
2. Pair the trackpad via Bluetooth (at this point there are two devices, at this point, I can see two devices in the Windows device manager: `Bluetooth HID Device`, and `Bluetooth XINPUT compatible input device`.
3. Install imbushuo driver `AmtPtpDevice.inf` again
4. Run `AmtPtpControlPanel.exe` and click `Install Driver`
5. Go to Windows Device Manager and select the two Bluetooth devices. Install the `Apple USB Precision Touchpad Device (user-mode)` for both via the `Update Driver` option.

When I tried to install the driver for either device, I received a `This device cannot start. (Code 10)` error. 

Is there any way
 to get it working on Bluetooth? Thank you

AnshulJ999, 2025-02-11:
Just want to follow this issue as I am planning on purchasing a Magic Trackpad myself as it seems the best option on Windows. 

Would like clarity if the new USB-C version supports Bluetooth via any driver or any method. Else I will try and find a Lightning version, which AFAIK does work with Bluetooth. Thanks.

vitoplantamura, 2025-02-12:
hi,

unfortunately currently as far as I know there is no way (open source, i.e. free) to use the USB-C MT2 in Windows via bluetooth. My version of the imbushuo driver only works via USB, as clearly specified in the project description and in the main README.

The problem is not only the cost of the EV certificate needed to sign the bluetooth driver: for example on my 2018 Asus laptop I have never managed to make my Lighting MT2 work with the imbushuo drivers via bluetooth... so when purchasing an MT2 to use it in Windows it would be useful to also keep this eventuality in mind.

Vito

AnshulJ999, 2025-02-17:
Hi @vitoplantamura , 

Thanks for the reply and your work on this project. I recently got a Magic Trackpad, Lightning version (2024 model, A1535 / MMMP3ZM/A). 

I tested various approaches with this myself to find the best one. And I agreed with your own assessment, as these are the best 2: 

1) Official Apple Drivers. However, instead of the 2021 drivers you mentioned, [I found these 2022 drivers ](http://swcdn.apple.com/content/downloads/03/60/041-96205/61hhcnj7q5dxosc171ytixty20vuqg0r0n/AppleBcUpdate.exe) on [this reddit post](https://www.reddit.com/r/bootcamp/comments/ygv1mh/any_way_to_get_magic_trackpad_2_working_on/)

2) Your solution built on top of imbusho's drivers. The control panel does give more flexibility over the feel. 

For now, I have installed 2 drivers: the official Bluetooth driver, and your solution. I felt the official Bluetooth driver works better for wireless use than imbusho's one. 

[image]

Overall, the stability of wired is definitely much greater than the Bluetooth one.

Now I have a few questions and feature requests. 

a) Haptic Feedback and Pressure: This might sound weird and unsupported, but is it possible to lower the pressure required to create a click, while maintaining a stronger haptic feedback? 

I am using my trackpad in a silicone cover, which slightly dampens the feedback. So I'd like to make the clicks easy, but still receive a moderate/high feedback. For now I'm testing all 3 options: Light, Medium, Firm, etc., and Light has a nice low-pressure click, but too soft feedback.

b) Since I have both imbushuo's driver installed and the official Bluetooth one, I am not really sure which one is being used wirelessly? Is there any way to verify? I know your control panel does not work in wireless mode (as expected), but otherwise it's hard to say which one is being used. 

c) 

[image]

Is there any better explanation of what these options do? I particularly disabled 'ignore input from finger used for force touch' and 'stop point when you lift a finger' because I faced weird results with these. But I'd like to better understand what these do. 

d) Quick suggestion, but would you consider crowdfunding the cost for signing a new Bluetooth driver that supports the new features? I understand it's an expensive undertaking, but I think there's a community that could get behind this hopefully. 

Thanks again! Really appreciate the time gone into all this.

vitoplantamura, 2025-02-20:
hi,

a) you can try to change the FeedbackClick and FeedbackRelease values in the registry, here:

Computer\HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\Windows NT\CurrentVersion\WUDF\Services\AmtPtpDeviceUsbUm\Parameters

After changing the values, you can unplug and replug the MT2 to see the difference.

A reference to the meaning of these values is here:

https://github.com/mwyborski/Linux-Magic-Trackpad-2-Driver/issues/28#issuecomment-451625504

b) you can check it in the Device Manager, in the MT2 properties.

c) yes, another person also expressed doubts about the clarity of these options... As soon as I have a chance, I will add a help text in the control panel.

d) unfortunately the main problem is not the cost of the EV certificate, as explained in my previous post, but the fact that the bluetooth version of the imbushuo driver does not work on some PCs (including mine)... unfortunately I do not have the time to debug and fix it.

Vito

sjain882, 2025-07-06:
Someone was able to solve your issue: https://github.com/vitoplantamura/MagicTrackpad2ForWindows/issues/20