vitoplantamura/MagicTrackpad2ForWindows #18: Polling Rate

dahkneth, 2025-03-10:
The polling rate of the Magic Trackpad 2 has been 95 Hz for sometime. With MacOS 16+ polling rate has increased for the Magic Trackpad. I've seen reports of polling rates up to 144 Hz and have tested myself with rates up to 125 Hz (my monitor can't go higher). While if I use it on Windows with this driver or the driver this is based on I can only reach 95 Hz polling rate.

95 Hz is a bit low and will feel sluggish on a modern monitor so have anyone investigated in how MacOS have unlocked/changed the polling rate when the device is connected to a Mac with MacOS 16+ and how that could be implemented in this driver?

vitoplantamura, 2025-03-10:
hi,

I need to take a deeper look at the imbushuo driver code to see where it is specified and if it can be modified. I hope I can do that soon.

The same question was asked here (without an answer): https://github.com/imbushuo/mac-precision-touchpad/issues/515

Vito

Jpe230, 2026-09-10:
I did some testing on a Lightning Magic Trackpad 2 (`05ac:0265`) on macOS and I think the higher “polling rate” people are seeing is probably **software interpolation/resampling rather than the trackpad actually sending HID reports at 120–250 Hz**.

I measured the raw HID input report cadence directly using `IOHIDDeviceRegisterInputReportCallback` on macOS, timestamping the incoming reports before any cursor/browser processing.

The result was consistently around **65–67 reports/sec**.

Interestingly, `hidutil dump` on the same Mac also shows:

```text
ReportInterval = 15000
InterpolateEvents = true
```

`15000 µs` corresponds to ~66.7 Hz, which matches the raw HID measurement almost exactly.

At the same time, I can observe roughly:

```text
Raw HID reports:        ~66 Hz
Browser pointer events: ~120 Hz
Desktop pointer events: ~240 Hz
```

So it looks like newer macOS versions are taking the relatively low-rate physical trackpad samples and **interpolating/resampling pointer motion to a higher cadence**, rather than unlocking a higher hardware polling mode.

I also looked through this driver's Magic Trackpad 2 initialization. For the Lightning model it enables the normal TYPE5 multitouch/Wellspring mode, but I couldn't find any explicit polling-rate setting. The Linux driver appears to use essentially the same `{ 0x02, 0x01 }` feature report to enable multitouch mode.

To reproduce the smoother macOS behavior, the interesting part may instead be implementing/interacting with **motion interpolation at the Windows Precision Touchpad / HID reporting layer**.

It would also explain reports of the apparent polling rate being tied to display refresh rate on macOS: the output cadence is being generated by the OS rather than directly reflecting the device's raw report rate.

vitoplantamura, 2026-09-12:
Thanks for your cool work!

It would be interesting to repeat your test with the USB-C MT, though...

Vito

startnw, 2026-10-04:
I have the same problelem polling rate is limited and it looks jagged on high refresh rate display. 
Hopefully someone will have the fix this issue soon.