vitoplantamura/MagicTrackpad2ForWindows #19: Issue installing driver error 10

Bmcaleese2023, 2025-03-12:
I originally found this https://github.com/maashrafh/MagicTrackpad2ForWindows which appears to be the same exact project as this. The only difference is that one has a side note at the bottom for updating the drivers manually in device manager since just clicking install doesnt fix the trackpad for usb-c. However when I try to do that I get the windows error 10 that says this 

[image]

Any idea on how to fix this issue?

vitoplantamura, 2025-03-12:
hi,

try installing the self-signed driver:

https://github.com/vitoplantamura/MagicTrackpad2ForWindows/issues/13#issuecomment-2553174594

Vito

tubeldelta, 2025-03-27:
Hi @vitoplantamura 

I have the same issue on an ARM Surface Laptop 7. After the error occurs, the device shows with an exclamation mark in Device Manager, and gestures don't work. 

Your self-signed driver can't easily be installed, either, as I get an error that the certificate chain could not be validated (even after installing the cert in `local machine` / `trusted root certification authorities`). I was able to install it by disabling driver signing completely via advanced startup options, but the result is the same as above - error 10, exclamation mark and no working gestures.

I'm pretty sure I've tried all possible orders / combinations, while making sure that the drivers is cleaned up in between attempts via forced uninstall.

I also have an AMD64 based machine, where everything worked perfectly on the first try, so I'm pretty sure the ARM drivers just don't work. 

Can I help you in any way to debug this and get it working?

vitoplantamura, 2025-03-28:
hi,

can you try to open `c:\windows\INF\setupapi.dev.log` with Notepad and to search for `"AmtPtpDevice"`? There should be one or more sections about your installation attempts. Can you post them here?

Vito

tubeldelta, 2025-03-28:
Here you go, I hope this helps: [setupapi.dev.log](https://github.com/user-attachments/files/19500004/setupapi.dev.log)

I'm not 100% sure what stage this is, but I think it was my last installation attempt before giving up. If you need more info or a clean attempt from the start, please let me know.

vitoplantamura, 2025-03-28:
unfortunately the log you attached refers to a successful attempt, probably the installation of the original imbushuo driver. I am interested in seeing everything that is logged when trying to install my self-signed driver.

Could you clean everything with `DriverStoreExplorer` and try to install the self-signed driver again? Please consider that the most recent events are at the end of `setupapi.dev.log`.

Also, as a further test I would ask you to import the certificate also in the **Trusted Publishers**, in addition to the Trusted Root Certification Authorities (where it should already be present).

Thank you, Vito

tubeldelta, 2025-03-29:
Hi @vitoplantamura 

Thanks, I've now done the following:

1. uninstall Magic Utilities (which I've been using in the meantime)
2. verify with `DriverStoreExplorer` that no drivers from Magic Utilities or imbushuo / Apple are left
3. [install the certificate under `Local Machine / Trusted Publishers`](https://github.com/user-attachments/assets/7d8bf229-c0be-42a8-8a48-3b9a0a6a335a) -> [this still caused an error](https://github.com/user-attachments/assets/1725e09f-2999-4d40-a978-96fd63bb21c4), but as I hadn't rebooted after disabling signed driver enforcement, I was able to install them anyway
4. [find the needed device in Device Manager](https://github.com/user-attachments/assets/55783d3f-a224-4846-b33c-d62090aa4060)
5. [install the driver](https://github.com/user-attachments/assets/c6e36fbd-12c6-49a2-b5b6-a576b2308da1) -> [this warning is shown](https://github.com/user-attachments/assets/f66fdbb2-7459-4123-aba7-262dc4593dd4)
6. [get error 10](https://github.com/user-attachments/assets/3c563016-053c-4d3b-ad62-0b7abbc99307)

The result is that [the device is shown with a warning](https://github.com/user-attachments/assets/557206b1-a8c6-44ea-8721-0272bfd3a989), and [shows error 10](https://github.com/user-attachments/assets/77051d54-213f-4f28-8d5d-a3097460f89b). Gestures don't work, and the control panel from your source has no effect (and shows an error when trying to install your DLL, which as I understand is expected, as the self-signed driver installs itself directly).

I've attached the full `setupapi.dev.log` from this attempt here: [setupapi.dev.log](https://github.com/user-attachments/files/19516568/setupapi.dev.log)

vitoplantamura, 2025-03-29:
Thank you for your detailed reporting.

I realize only now that we were referring to different procedures 😀 Thanks for your patience.

To install the self signed driver (in order to generate useful information in `setupapi.dev.log`) you need to follow these steps:

1) clean everything with `DriverStoreExplorer`.

2) download the zip file from here: https://github.com/vitoplantamura/MagicTrackpad2ForWindows/issues/13#issuecomment-2553174594

3) install the certificate in the zip file in Trusted Root Certification Authorities and Trusted Publishers.

4) right click on the INF file in the zip file and click Install.

5) collect the text that is logged in `setupapi.dev.log`.

Thank you again for your patience,

Vito

tubeldelta, 2025-03-29:
@vitoplantamura no worries, and thanks for your help, I appreciate it! 🙂

The steps you describe are exactly what I did in my previous post, and what the output log refers to. I saw now that I didn't describe well what I did in my third step. I first installed the certificate and then the driver, and still got an error.

vitoplantamura, 2025-03-30:
ah ok! 😀

Googling for the error message in your `setupapi.dev.log` I found this issue:

https://github.com/VirtualDrivers/Virtual-Display-Driver/issues/197

In this project, the installation procedure is the same as our self signed driver.

Unfortunately I think there is little we can do: they write: "Custom root certs no longer are checked in windows arm64".

I'll do some more research: if I find anything else relevant I'll let you know in this issue.

Thank you, Vito

tubeldelta, 2025-03-30:
@vitoplantamura good find, thanks!

Do you think the "error 10" is related to the driver signing at all? I'm asking because I was able to install the unsigned driver by disabling enforced driver signing in advanced startup options, but that didn't solve this error.

vitoplantamura, 2025-03-31:
good question: since DSE is disabled, the problem should be elsewhere.

To verify this hypothesis, just see if the driver DLL is loaded: with DSE
disabled, plug in the MT2 and with Resource Monitor look for
AmtPtpDeviceUsbUm.dll. If the DLL is loaded, the problem is in the code.

Link (second answer):
https://superuser.com/questions/117902/find-out-which-process-is-locking-a-file-or-folder-in-windows

Thank you, Vito

tubeldelta, 2025-04-01:
Hi @vitoplantamura 

Thanks, I've tried that now as follows:

1. uninstall Magic Utilities
2. install your self-signed driver with DSE off (warning is shown but installation can be completed)
3. change to the `Bingxing Wang` driver for device `\VID_05AC&PID_0324&MI_01&Col02\7&1797f234&0&0001`
4. check Resource Monitor
5. replug the MT2 just to be sure

[It doesn't look like it's being loaded](https://github.com/user-attachments/assets/712b70da-eddb-427a-a49f-af7b7f6689b9)

tubeldelta, 2025-04-02:
Glad to hear there was some insight to be gained :)

Yes, I can make changes on the ARM Laptop with the USB C Trackpad, and they are applied (I disabled the "ignore input from the finger used to click the force touch button" option, for example).

konradzuse, 2025-10-22:
i'm also having issues with a Surface Pro 11. Are you talking here about USB or Bluetooth? I'm able to get it working on USB, but Bluetooth HID device gives me a code 10.

vitoplantamura, 2026-02-24:
hi,

I'll post this message in multiple relevant issues in the repo.

So at the beginning of this year, I decided to purchase an EV certificate to properly sign the driver. I was tired of seeing people resorting to the wildest hacks to get the MT2 to work via Bluetooth 😀

I need help testing the driver, especially for hardware I don't have access to, specifically Windows on ARM and the Magic Trackpad 2 USBC.

This signed version supports all the existing features of this project, plus **Bluetooth** and **battery level reading**.

Link: [MT2W11-20260223-MSSigned.zip](https://github.com/user-attachments/files/25516148/MT2FW11-20260223-MSSigned.zip)

Important:

1) Currently, **only Windows 11 is supported**. Let me know if support for Windows 10 is also desired. NOTE: Windows 10 is not supported out of the box due to issues (not immediately fixable) with the INF validation during signing.

2) The driver can be installed by right-clicking the INF file and selecting "Install". I strongly suggest **deleting any previous installed drivers** before proceeding: I use "DriverStore Explorer" for that.

Thanks,

Vito