vitoplantamura/MagicTrackpad2ForWindows #22: Click pressure sensitivity

alexander-denev, 2025-06-16:
This project is awesome.
Even more awesome would be to add an option for setting the required pressure for the magic trackpad to register a click.

I haven't read the code of this repo or any other connected repos for the drivers, so I don't whether the software or hardware supports this.

It would be appreciated if someone in the know could tell me if this can be coded in, so that I might look into doing so.
Thanks for reading!

vitoplantamura, 2025-06-19:
hi,

sorry for the late reply.

You can indirectly set the amount of pressure required to register a click by changing the haptic feedback options in the Control Panel.

If you want to decouple the two settings (for example silent clicking with more force to actuate the click) you can try to change the two values ​​directly in the registry: https://github.com/vitoplantamura/MagicTrackpad2ForWindows/issues/17#issuecomment-2670910675 (point a).

It's a matter of doing some tests to understand the meaning of the various hexadecimal values.

Let me know if you discover anything,

Vito

bradspage, 2026-09-30:
Did you ever figure out if any of those registry keys give it a lighter pressure to click? My fingers hurt 😂